arduino/ArduinoCore-avr #339: analogWrite - PWM Problem in IDE-Version 1.8.10

Nnambles, 2020-05-04:
There is a Bug in Version 1.8.10 of the Arduino IDE using analogWrite() for PWM on Arduino Mega 2560. If i use analogWrite() with any value for the pwm-Pins once in the loop()-Method everything works fine. If I put a second call of analogWrite() with the same pin and any value that is not 0 or 255, nothing happens. The loop-Method keeps on working, but the PWM-Pin used in analogWrite() shows an continous output of 0V. In Version 1.8.9 the same programm works fine.

WestfW, 2020-05-05:
I've created a minimalized source tree and build scripts for working on this in https://github.com/WestfW/Duino-hacks/tree/master/pwmbug
These:

1. bypass the IDE build process
1. include a minimalized source tree that produces identical code to an IDE build
1. have several build scripts for comparing object files.

(no actual results yet.)

WestfW, 2020-05-07:
I've further consolidated the minimalized source, end up with a "consolidate" version that has one .cpp file and one .c file, each already pre-processed.  (main.cpp has the main.cpp and sketch code, wiring_all.c has the necessary parts of wiring.c, wiring_digital.c, wiring_analog.c)
And I checked the .hex file to confirm it exhibits the bug on live hardware.
github repository updated, and minimal form attached...
[pwmbug.zip](https://github.com/arduino/ArduinoCore-avr/files/4591307/pwmbug.zip)

It's still pretty mysterious.  For example, it will start working if the code for turnOffPWM() is commented out.

WestfW, 2020-05-08:
It certainly looks like an off-by-one error in calculating the start of the jump table, as theorized in https://github.com/arduino/ArduinoCore-avr/issues/339#issuecomment-596062769

What's weird is that it doesn't seem to happen for the non-lto case, doesn't happen when there is only a single analogWrite() call, and **(NEW DATA!) doesn't happen if the -relax option is removed from the link command.**

**-relax** causes certain long jumps/calls to be shortened to rjmp/rcall.  Perhaps the intermediate representations passed to the linker with -flto are missing information as to locations that need "fixed up" after "relaxing" ?

This is pretty much as far as I can go.  I have no knowledge or understanding of avr-gcc internal operations...

matthijskooijman, 2020-05-19:
> What's weird is that it doesn't seem to happen for the non-lto case, doesn't happen when there is only a single analogWrite() call, and (NEW DATA!) doesn't happen if the -relax option is removed from the link command.

Ah, that is awesome new info, since that gives a very strong indication of what is happening here (jump tables getting messed up by jump relaxation).

I had hoped that this extra info might turn up an existing bug report about this, but no such luck it seems.

The minimized example also looks good. Not sure if it is minimal enough yet for a gcc bugreport, I suspect that there are still some definitions and declarations that can be removed? Also, it would be good if the generated assembly is simple enough so the problem can be shown based on that, rather than having to run the code, but that might be tricky and/or cause the problem to disappear...

One thing that could maybe help is to pass `--param case-values-threshold=1` to gcc, to make it generate a jump table for all switches (i.e. set the minimum number of case labels for using a jumptable to 1). Maybe with that, a minimal example could be constructed rather than basing on the existing example, or the existing example could be further reduced?

Can we test this on newer gcc versions (or did we already)? It seems Arduino now ships 7.3.0, not sure if there are any versions available in staging or elsewhere? Debian only has 5.4.0 it seems...

hb9tst, 2020-05-28:
Just discovered this issue after trying to figure out for hours what was wrong with PWM. It is a very annoying bug... especially that I knew that all this was working in the past!

From what I could observe in my tests (on two Mega2560 R3 boards), as soon as there is a second call to analogWrite() that is compiled _(unless it is **identical** to the first one),_ all calls are not working at all, any only a flat 0V signal is observed with the oscilloscope on the corresponding output pin.

This code works fine:

``` C++
void setup() {
  pinMode(2, OUTPUT);
  analogWrite(2, 42);
}
```

This works too:

``` C++
void setup() {
  pinMode(2, OUTPUT);
  analogWrite(2, 42);
  analogWrite(2, 42);
}
```

This **does not** work:

``` C++
void setup() {
  pinMode(2, OUTPUT);
  analogWrite(2, 42);
  analogWrite(2, 84);
}
```

This works, as the second call is optimized out:

``` C++
void setup() {
  pinMode(2, OUTPUT);
  analogWrite(2, 42);
  if (1 == 0) {
    analogWrite(2, 84);
  }
}
```

This **does not** work as the second call is **not** optimized out by the compiler:

``` C++

int x = 12;

void setup() {
  pinMode(2, OUTPUT);
  analogWrite(2, 42);
  if (x == 0) {
    analogWrite(2, 84);
  }
}
```

What is surprising is that the second call to `analogWrite` never happens, because `x` is not zero. However, the fact that the call is present seems to be sufficient for the compiler to generate buggy code. I confirmed using `Serial.println()` that the code in the `if` statement is never executed.

In all cases my `loop()` was empty (or contained calls to `Serial.println()`, which didn't influence the behaviour of `analogWrite()`. Unfortunately I don't have time to compare the generated assembly code, also because my knowledge of AVR assembly is very superficial.

facchinm, 2020-06-22:
Should be fixed by avr core 1.8.3 + IDE 1.8.13 ; updating to the new IDE is important since it ships with [this](https://github.com/arduino/arduino-cli/commit/3b18571924405b9175cd914d4ef20b07618b92d4) patch (on the tooling side).
If people previously experiencing the problem could report here if they still have troubles it would be great.
In the meantime, I'm closing this as fixed.

closed 2020-06-22 by facchinm

matthijskooijman, 2020-06-22:
For completeness, this is not actually *fixed* (the bug still exists in gcc), we just removed the circumstances that trigger this bug. Small (but IMHO relevant) difference. Any third party cores that would add `--relax` manually would still be affected too.