arduino/ArduinoCore-avr #342: Wire Slave: arbitrary user code is called from ISR

greyltc, 2020-05-16:
When a user is using the Wire library in slave mode, they generally need to register a function to handle requests for data from the master. They write some function, which must contain `Wire.write` and register it like this: `Wire.onRequest(someFunction);`.

The issue here is that `someFunction` gets called (via `twi_onSlaveTransmit`) right in the slave's interrupt service routine: https://github.com/arduino/ArduinoCore-avr/blob/58081c05e548560d3a60050bbf260ebec5d1e867/libraries/Wire/src/utility/twi.c#L528

First of all, the documentation for `Wire.onRequest` [is truly tragic](https://www.arduino.cc/en/Reference/WireOnRequest). It doesn't even tell the user they must make a call to `Wire.write` or else [the wire library will decide for them](https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/utility/twi.c#L530-L533) to send `0x00` to the master, and it doesn't warn the user that they're writing code for inside an ISR.

Secondly, this architecture raises all kinds of potential usability issues. I think quite often a slave will need to do something much more complicated to answer a master's request than what should/can be done in an ISR. As I understand the code (please someone correct me if I'm wrong), it's impossible to answer the master outside of the call from the ISR., it's impossible to service a master's request with anything other than code run from the ISR. What if the slave needs the `delay()` function in forming its answer for the master? Impossible. What if the slave needs to fetch an answer for the master over SPI? *Maybe* that's possible, but it sure seems like a terrible idea to do that from inside an ISR.

I noticed a stub:

https://github.com/arduino/ArduinoCore-avr/blob/58081c05e548560d3a60050bbf260ebec5d1e867/libraries/Wire/src/Wire.cpp#L271-L274

Maybe the point of that was to help solve this problem?

Would you consider accepting a PR I might make so that a slave can form an answer to a master outside of the ISR?

greyltc, 2020-05-18:
I think [my issue#342 branch](https://github.com/greyltc/ArduinoCore-avr/tree/issue%23342) solves this. Need to do a bit more testing to check if this is backwards compatible with the Wire library API before this change.

Ragebone, 2021-11-17:
@greyltc 

> As I understand the code (please someone correct me if I'm wrong), it's impossible to answer the master outside of the call from the ISR.

I understand the current implementation like that as well. 

> I think [my issue#342 branch](https://github.com/greyltc/ArduinoCore-avr/tree/issue%23342) solves this. Need to do a bit more testing to check if this is backwards compatible with the Wire library API before this change.

Have you had any progress on this? 
After a brief look, i think this should be backwards compatible since your additional handling is using the previously unused return-value of [twi_onSlaveTransmit](https://github.com/greyltc/ArduinoCore-avr/blob/35e334c51e23588d077e909cdfa77fff4e104c48/libraries/Wire/src/utility/twi.c#L647)

Do i understand correctly that you  [disable TWI Interrupts](https://github.com/greyltc/ArduinoCore-avr/blob/35e334c51e23588d077e909cdfa77fff4e104c48/libraries/Wire/src/utility/twi.c#L657) to cause clock-stretching?
And then enable them again on flushing?

One of your [commits](https://github.com/greyltc/ArduinoCore-avr/commit/35e334c51e23588d077e909cdfa77fff4e104c48#diff-3bb417099563148b00d182b14fbd787c3a6ab5c6f5875054291cbf75fa7892b1R649) for reference.

> I think quite often a slave will need to do something much more complicated to answer a master's request than what should/can be done in an ISR.
> [...]
> What if the slave needs the `delay()` function in forming its answer for the master? Impossible.

I kind of disagree because i had issues with "excessive" clock-stretching on an Arduino Uno.
I was implementing the i2c protocol of an IR35201 which only requires to answer a single Byte write with a single Byte from a Register. 
All the needed information is already present in the Wire.onReceive() registered method.
So Wire.write(outByte) was the only operation i was needing in the onRequest() registered method.
But the call to an empty  twi_onSlaveTransmit() was enough to cause the last two Data-Bytes to be stretched consistently. 
To me it was an issue because the used Master would error the moment clock-stretching occurred.
After a lot of digging, i was only able to reduce it to sporadic clock-stretching.
Because of that, i am under the impression that as long as the ISR has not finished, the TWI Hardware will clock-stretch. 
S that makes me think that you can do what ever you want in the twi_onSlaveTransmit() method. 

>[...] , but it sure seems like a terrible idea to do that from inside an ISR.

I have no idea if that is a bad idea or not.