arduino/ArduinoCore-avr #346: shiftin() bad code size and speed

apollo2mond, 2020-06-11:
Hi,
shiftin() is horrible inefficient especially for small machines because of the use of "dynamic shifts" like (1<<i). and shiftout() needs a cleanup.
I propose a replacement with the functions below which have better code size and speed based on the use of "static shifts"! Keep going!
```
uint8_t shiftIn(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder) {
	uint8_t val = 0;

	for (uint8_t i = 0; i < 8; ++i) {
		digitalWrite(clockPin, HIGH);
		if (bitOrder == LSBFIRST)
			val = (val>>1) | (digitalRead(dataPin)<<7);
		else
			val = (val<<1) | digitalRead(dataPin);
		digitalWrite(clockPin, LOW);
	}
	return val;
}

void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val) {
	for (uint8_t i = 0; i != 8; i++)  {
		if (bitOrder == LSBFIRST)
			digitalWrite(dataPin, val & 0x01), val >>= 1;
		else
			digitalWrite(dataPin, !!(val & 0x80)), val <<= 1;
			
		digitalWrite(clockPin, HIGH);
		digitalWrite(clockPin, LOW);		
	}
}
```

nerdralph, 2020-07-11:
Counting down from 8 to 0 usually results in smaller and faster code.
http://nerdralph.blogspot.com/2015/03/fastest-avr-software-spi-in-west.html

I believe my picoCore has the smallest & fastest shiftIn and shiftOut.
https://github.com/nerdralph/picoCore/blob/master/avr/cores/picocore/wiring.c

RobTillaart, 2020-12-05:
@permaBox 
You might have a look at https://github.com/RobTillaart/FastShiftIn