arduino/ArduinoCore-avr #349: Creating Readme.md file

AshutoshPandey123456, 2020-06-21:
Unlike the samd core repo https://github.com/arduino/ArduinoCore-samd/blob/master/README.md 

This one does not seem to have a readme file. A readme with details about the contents of the repository and structure along with a link on how to install it in Arduino CLI would be helpful.

Can I work on this and submit a PR?

AshutoshPandey123456, 2020-06-22:
Went ahead and created a PR: https://github.com/arduino/ArduinoCore-avr/pull/350

per1234, 2025-09-04:
@ritesh006 please take a bit more time to research before commenting on issues. If you had done that, you would see that #413 does not at all resolve this proposal:

>A readme with details about the contents of the repository and structure along with a link on how to install it in Arduino CLI

Do you see any of that in https://github.com/arduino/ArduinoCore-avr/blob/master/README.md?

I already explained all this in my comment on the linked PR https://github.com/arduino/ArduinoCore-avr/pull/350#issuecomment-1003576037

I appreciate that you are trying to help with the maintenance of the issue tracker, but please understand that "noise" like this comment you made is the opposite of helpful. There are tens of thousands of issues across hundreds of Arduino's repositories. The maintainers of those repositories must monitor all those conversations for comments that make valuable technical contributions. So even just a little bit of "noise" here and there turns into a massive number of notifications and distractions for us.

When an issue becomes a source of "noise", I am forced to lock it. That makes it so that contributors who do have something important to add can not do so.

If you would like to contribute to the Arduino project in an environment where extensive discussion is welcome, please feel welcome to participate in Arduino Forum:

https://forum.arduino.cc/

It is a great place to share knowledge, assist members of the Arduino community with their projects, and to discuss any subjects related to embedded systems.

ritesh006, 2025-09-06:
Thanks for clarifying, @per1234. Understood—I’ll make sure to check the existing discussion more carefully before commenting next time. Appreciate your explanation. 👍