arduino/ArduinoCore-avr #365: USB_PRODUCT macro makes custom USB Device names convoluted to modify

midiclub604, 2020-10-05:
https://github.com/arduino/ArduinoCore-avr/blob/3055c1efa3c6980c864f661e6c8cc5d5ac773af4/cores/arduino/USBCore.cpp#L44-L49

Hey team, I'm very novice C++ programmer, but this block of the USBcore.cpp file code seems to make it virtually impossible to make a custom USB Device name by any other means except modifying the boards.txt file or modifying the USBcore.cpp core file. The const u8 STRING_PRODUCT[] array is initialized here internally only.

If someone is making a project for an arduino game controller, midi controller, or anything that is built on top of the pluggableUSB library, their arduino USB device will always show up as the name of the board on their computer's USB device list, ex. "Arduino Micro"

It's easy enough to modify the name on the boards.txt file, however it might be a complicated extra set of steps for some users of a diy project.  Teensy USB MIDI device names can be quickly modified in a config.h file in the project directory for example. I'm not sure what the best approach here is, but could this issue be addressed somehow? Or if there is a way to do this without modifying the core, boards, or platform file please let me know.  Thank you!

facchinm, 2020-10-05:
Hi @midiclub604 ,
the suggested way to modify the name is creating a derived core and override the name from there. You can take inspiration from https://github.com/webusb/arduino/tree/gh-pages/hardware/webusb/avr (which is very thing and easy to maintain).

closed 2020-10-05 by facchinm

midiclub604, 2020-10-05:
Awesome, thanks for the tip!