arduino/ArduinoCore-avr #366: .hex file with bootloader is incorrect for ATmega2560 targets (Arduino Mega2560)

WestfW, 2020-10-08:
the .ino.with_bootloader.hex file produced for ATmega2560 targets (and prrobably any AVR target with more than 64k flash) is missing the  "extended segment address" entry, so the bootloader ends up positioned at 64k-8k, instead of 256k-8k.

Note the VMA address (0xe000), which should be 0x3e00)

~~~~
avr-objdump -x /tmp/Arduino1.8.13Build/Blink.ino.with_bootloader.hex   

/tmp/Arduino1.8.13Build/Blink.ino.with_bootloader.hex:     file format ihex
/tmp/Arduino1.8.13Build/Blink.ino.with_bootloader.hex
architecture: UNKNOWN!, flags 0x00000000:

start address 0x00000000

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .sec1         00000600  00000000  00000000  00000010  2**0
                  CONTENTS, ALLOC, LOAD
  1 .sec2         00001d1e  0000e000  0000e000  00001090  2**0
                  CONTENTS, ALLOC, LOAD
~~~~

And here's that section of the .hex file:

~~~~
:1005C000F894EC91E22BEC938FBFC0E0D0E081E097
:1005D0000E942D010E94DA0180E00E942D010E94FC
:1005E000DA012097A1F30E940000F1CFEE0FFF1F68
:1005F000881F8BBF0790F691E02D1994F894FFCFD8      ;; last code from sketch
:10E000000D9489F10D94B2F10D94B2F10D94B2F129     ;; first code from bootloader
:10E010000D94B2F10D94B2F10D94B2F10D94B2F1F0
:10E020000D94B2F10D94B2F10D94B2F10D94B2F1E0
~~~~

It should look like:

~~~~
:1005C000F894EC91E22BEC938FBFC0E0D0E081E097
:1005D0000E942D010E94DA0180E00E942D010E94FC
:1005E000DA012097A1F30E940000F1CFEE0FFF1F68
:1005F000881F8BBF0790F691E02D1994F894FFCFD8      ;; last code from sketch
:020000023000CC
:10E000000D9489F10D94B2F10D94B2F10D94B2F129     ;; first code from bootloader
:10E010000D94B2F10D94B2F10D94B2F10D94B2F1F0
:10E020000D94B2F10D94B2F10D94B2F10D94B2F1E0
~~~~

matthijskooijman, 2020-10-08:
I believe this is related to, or probably even the same issue as https://github.com/arduino/arduino-cli/issues/1009

WestfW, 2020-10-09:
Could be.  I did not use a "large" sketch for the example problem (it was Blink.ino...)
If the sketch were large, the bootloader and sketch could collide, with different end symptoms.

facchinm, 2020-10-09:
Hi Bill,
it's indeed the same problem since the bootloader section address in the Mega is over 64K into the flash space.
On the other report (SAMD), the bootloader sits at 0x0000 and the issue appears when the sketch becomes bigger than 0xFFFF .
I prepared a patch for `arduino-cli` since the builder is based on it. If you don't want to install a go toolchain I can send you a binary of the builder for any platform so you can test if the fix is working.

closed 2020-11-03 by cmaglie