arduino/ArduinoCore-avr #371: random(x*7)-function only gives smaller multiples of 7 on first call

ShantySuffisance, 2020-11-30:
Hi, 

```
int n;

void setup() {
  Serial.begin(9600);
  randomSeed(analogRead(A0));
  n = random(21);
  Serial.println(n); 
}
void loop() {
}
```
only outputs numbers from the set:  {0,7,21}. All multiples of seven give back the smaller multiples of seven on the first call. I tested on arduino-hardware and on tinkercad (simulation). 

incerely yours
shanty

RobTillaart, 2020-12-05:
Interesting issue

```cpp
int count[21];

int n;

void setup() {
  Serial.begin(115200);
  for (int seed = 0; seed < 30000; ++seed) {
    randomSeed(seed);
    n = random(21);
    // Serial.println(n);
    count[n]++;
  }
  for (int i = 0; i < 21; i++)
  {
    Serial.print(i);
    Serial.print("\t");
    Serial.println(count[i]);
  }
}

void loop() {
}
```

for the first 30K seeds 7x has a significant higher chance.

RobTillaart, 2020-12-05:
Some further testing shows that the 7-bias exist if **maxValue** is a multiple of 7

```cpp
int count[350];

int n;

void setup()
{
  Serial.begin(115200);
  for (int mx = 7; mx < 350; mx += 7)
  {
    for (int i = 0; i < mx; i++)
    {
      count[i] = 0;
    }

    for (int seed = 0; seed < 1000; ++seed)
    {
      randomSeed(seed);
      n = random(mx);
      // Serial.println(n);
      count[n]++;
    }

    Serial.println();
    Serial.println(mx);
    for (int i = 0; i < mx; i++)
    {
      if (count[i] != 0)
      {
        Serial.print(i);
        Serial.print("\t");
        Serial.println(count[i]);
      }
    }
  }

}

void loop() {
}
```

The problem gets even worse when **maxValue** = 49, 98, 147, 196, 245, 294, etc ==> have a 7^2 in it.
A test with 343 which is 7^3 was even more biased (bias cubed).

RobTillaart, 2020-12-05:
The patch in **WMath.cpp** proposed above does help a bit but there still is a serious bias.
For **maxValue** == 343 the histogram is plain wrong. 

Solution could be using a better random generator like **Marsaglia()**

**WMath.cpp**
```cpp
// DO NOT USE STD RANDOM
// extern "C" {
  // #include "stdlib.h"
// }

///////////////////////////////////////////////////////////////////////////
//
// An example of a simple pseudo-random number generator is the 
// Multiply-with-carry method invented by George Marsaglia.
// it has two initializers (not zero) which can be changed 
// to seed the generator.
//
unsigned long m_w = 1;
unsigned long m_z = 2; 

unsigned long Marsaglia()
{
    m_z = 36969L * (m_z & 65535L) + (m_z >> 16);
    m_w = 18000L * (m_w & 65535L) + (m_w >> 16);
    return (m_z << 16) + m_w;  /* 32-bit result */
} 

void randomSeed(unsigned long seed)
{
  if (seed != 0) m_w = seed;
}

long random()
{
  return Marsaglia() & 0x7FFFFFFF;
}

// other random calls are identical
```

ace-dent, 2021-06-20:
I'd humbly suggest not rushing to replace the 'Lehmer' RNG, which has been good enough for many years.
It'd be prudent to determine whether this is a change to the implementation somehow causing a bug now.
There are numerous algorithms that should be considered for a replacement RNG, and there must be a balance of 'quality' -vs- compiled size and RAM overhead.

RobTillaart, 2021-06-21:
@ace-dent

As the proof is in the pudding test I wrote a small comparison sketch, ran on an UNO - IDE 1.8.13
It does a loop of 1000 calls to random and assign them to a volatile variable 
so the compiler is not allowed to optimize the call.

```cpp
//    FILE: rnd_size.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: compare build-in random versus Marsaglia
//    DATE: 2021-06-21

unsigned long m_w = 1;
unsigned long m_z = 2; 

unsigned long Marsaglia()
{
    m_z = 36969L * (m_z & 65535L) + (m_z >> 16);
    m_w = 18000L * (m_w & 65535L) + (m_w >> 16);
    return (m_z << 16) + m_w;  // 32-bit result
} 

void randomSeedM(unsigned long seed)
{
  if (seed != 0) m_w = seed;  // just change one to keep interface randomSeed identical 
}

long randomM()
{
  return Marsaglia() & 0x7FFFFFFF;
}

uint32_t start, stop;
volatile uint32_t x  = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  start = micros();
  for (int i = 0; i < 1000; i++)
  {
    x = random();  // <<<<<< replace with randomM
  }
  stop = micros();

  Serial.println(stop - start);
}

void loop()
{
}

// -- END OF FILE --
```

**build in random()**
```
Sketch uses 2294 bytes (7%) of program storage space. Maximum is 32256 bytes.
Global variables use 232 bytes (11%) of dynamic memory, leaving 1816 bytes for local variables. Maximum is 2048 bytes.

OUTPUT: 51276
```

**Marsaglia()**   // replace random with randomM in code above
```
Sketch uses 2012 bytes (6%) of program storage space. Maximum is 32256 bytes.
Global variables use 236 bytes (11%) of dynamic memory, leaving 1812 bytes for local variables. Maximum is 2048 bytes.

OUTPUT: 8792
```

So it uses 
- 182 bytes less progmem,  
- 4 bytes **more** RAM and 
- is more than 5.8x faster  (in my test)

So unless you cannot spare the 4 bytes RAM,  **Marsaglia()** is an improvement in terms of size and performance. I do not know about the quality of the randomness of the **Marsaglia()** function, definitely not an expert in that science , but Mr. Marsaglia has serious credentials, so I expect that will be OK too. 

- https://en.wikipedia.org/wiki/George_Marsaglia

----
of course the interface for random(n) and random(n, m) should be tested likewise.

RobTillaart, 2021-06-21:
tested the other two with a naive implementation

```cpp
//    FILE: rnd_size.ino
//  AUTHOR: Rob Tillaart
// PURPOSE: compare build-in random versus Marsaglia
//    DATE: 2021-06-21

unsigned long m_w = 1;
unsigned long m_z = 2; 

unsigned long Marsaglia()
{
    m_z = 36969L * (m_z & 65535L) + (m_z >> 16);
    m_w = 18000L * (m_w & 65535L) + (m_w >> 16);
    return (m_z << 16) + m_w;  // 32-bit result
} 

void randomSeedM(unsigned long seed)
{
  if (seed != 0) m_w = seed;  // just change one to keep interface randomSeed identical 
}

long randomM()
{
  return Marsaglia() & 0x7FFFFFFF;
}

long randomM(uint32_t n)
{
  return randomM() % n;        // behaves worse for n > sqrt(max long)   !!!!
}

long randomM(uint32_t n, uint32_t m)
{
  return n + randomM(m - n);
}

uint32_t start, stop;
volatile uint32_t x  = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  start = micros();
  for (int i = 0; i < 1000; i++)
  {
    x = randomM(100, 200);  // <<<<<< replace with randomM
  }
  stop = micros();

  Serial.println(stop - start);
}

void loop()
{
}

// -- END OF FILE --
```

output of my test runs in a table 
```

                      PROGMEM       RAM        TIME
random()              2294          232        51276
randomM()             2012          236        8792
                   -----------------------------------
                      -282          +4         5.8 x

random(1000):         2314          232        90540
randomM(1000):        2024          236        47044
                   -----------------------------------
                      -290          +4         1.9 x

random(100, 200):     2320          232        91088
randomM(100, 200):    2032          236        47592
                   -----------------------------------
                      -288          +4         1.9 x

```

I do not know if there is a version of **Marsaglia()** that uses less RAM.

RobTillaart, 2021-06-21:
looked at the arduino code in Wmath.cpp, which is equally naive for larger params.
```cpp
long random(long howbig)
{
  if (howbig == 0) {
    return 0;
  }
  return random() % howbig;   // <<<<<<<<<<<<<<
}

long random(long howsmall, long howbig)
{
  if (howsmall >= howbig) {
    return howsmall;
  }
  long diff = howbig - howsmall;
  return random(diff) + howsmall;
}
```