arduino/ArduinoCore-avr #380: Wire example i2c_scanner starts at invalid i2c address

bperrybap, 2020-12-20:
The i2c_scanner example included with the Wire library starts its scan at i2c address 1.
There is no such i2c slave address.
The addresses 0 to 7 are not valid i2c slave addresses and are reserved because those addresses overlap with the timing for other i2c functions.
If you want details you can see here: https://www.i2c-bus.org/addressing/
Nick Gammon updated his version of the i2c_scanner code on his site quite a while back:
https://gammon.com.au/i2c

While not a factor for the AVR h/w, starting in the reserved space can cause a lockup on other platforms that use different i2c h/w.

--------------------

The fix is trival. Just start the scan at address 8.
It might be  helpful to put in a comment about it to explain to people why the scan starts at 8
i.e. change this: 
` for (byte address = 1; address < 127; ++address) {`

to something like:
```
// i2c addresses 0-7 are not valid slave addresses and are reserved for special purposes
// so start the scan at address 8
// for more on i2c addressing: // https://www.i2c-bus.org/addressing/
for (byte address = 8; address < 127; ++address) {
```

per1234, 2020-12-20:
Is this the same as https://github.com/arduino/ArduinoCore-avr/issues/319?

bperrybap, 2020-12-20:
Uh, yeah! Ooops.
Even worse, I'm the one that created the #319 issue.
I'm a total bonehead.
I did look to see if there was an issue for this and didn't see it before I created this one as I thought I had reported this before and thought it had already been fixed.
I guess this one can be closed as a duplicate.