arduino/ArduinoCore-avr #389: Setting the board serial number explicitly in USB HID API

abratchik, 2021-01-22:
Current implementation of USB HID library generates the Serial No automatically by adding the descriptor size to the "HIDAA" string. This approach is ok as a default but in some use cases it is necessary to set a specific serial number to the board, which will not change when the USB HID descriptor is updated. 
Besides, the serial number is utilized in the naming of the Arduino port so changing it dynamically may lead to the loss of connectivity to the board after the sketch upload.
Instead, a new method setSerial() is proposed, which can take any string stored in PROGMEM.

matthijskooijman, 2021-01-22:
Sounds like a good feature, I've thought about this before (for a project where machines have an actual serial number written to EEPROM, where it would be good to actually use that serial on USB as well). However, I wonder: How is the timing of this? Typically, USB is initialized directly on startup, even before `setup()` runs. Won't a `setSerial()` method be called too late to still influence the USB descriptor?

abratchik, 2021-01-22:
Timing is fine since all the string  descriptors can be overridden through PluggableUSB (it is called first in the USBCore).