arduino/ArduinoCore-avr #409: atmega16u2 baud rate work 1/8 of setting

DMohammed, 2021-04-28:
I had an Arduino mega, atmega16u2 with a faulty chip, therefore, i swapped the IC for a new one, the problem is after the fix i can only not program the chip i tried the following :
1_using flip bootloader that was provided on the website
2_ I changed bootloader :
3_diffrent baud rate speed 
4_change crystal
5_diffrent DFU files
none of the above solutions did work for me, then i proceeded to try another way that i found out with any baud rate speed i receive 1/8 of the initial speed but no luck there too all DFU file had the same problem, i tried the same procedure with normal Arduino and i had no problems

any help would be appreciated

matthijskooijman, 2021-04-28:
Sounds like you need to program the fuses on your 16u2 chip. By default, they run with CKDIV8 fuse set (and probably also configured for internal oscillator), which typically explains a x8 or x16 slowdown of the CPU clock.

DMohammed, 2021-04-28:
Thank you
fixed based on the following link 
https://github.com/nickgammon/arduino_sketches

closed 2021-04-28 by DMohammed