arduino/ArduinoCore-avr #418: analog_write(): wrong management of input val

Addictek, 2021-07-08:
Hello, the analogWrite() function takes the pin number which is uint8_t type and the target value of pwm duty cycle which is int type. Using 16 bit type for the target value is correct since some of the timers (for example TIMER-1 on 328p) have 16 bits resolution. Anyway, at the beginning of the function there is a check if the value is 255.

...
  else if (val == 255)
  {
    digitalWrite(pin, HIGH);
  }
....

In that case, the code sets the pin output steady high. This seems to be incorrect for pins supporting 16 bit timers. Nevertheless, if the input value is for example 258, the check is skept and the value is assigned to the timer register (es. ORC1A/B).
This is apparently inconsistent, maybe the code should check ">=255" instead of "==255"? Or the check should be against 65535 for pins supporting 16 bit resolution? Finally, why "val" parameter is int type instead of uint16_t?

matthijskooijman, 2021-07-08:
The analogWrite API is designed to only support 8-bit values. Any 16-bit timers are, IIRC, configured in 8-bit mode, so AFAIK there is currently no way to use this extra resolution. If you call it with > 255, I guess the behavior is mostly undefined...

Note that there are some cores that do support higher resolution (IIRC STM32, maybe SAM or SAMD too, but not AVR AFAIR), using an additional `analogWriteResolution` function, which globally sets the number of bits that analogWrite accepts. Not a great API, IMHO, but on those cores, values are scaled inside analogWrite before any special handling is applied I think.

Addictek, 2021-07-08:
ok, a bit misleading but make sense. So if the API is designed for 8-bit, why the 16-bit type for "val" parameter? Is that to keep the doors open for future improvements?

matthijskooijman, 2021-07-08:
I do not really know, but my suspicion is that this just was not really well-considered, or maybe the original design just used "int" everywhere a number was used to keep things simple (and not confuse users with overly complex looking `uint8_t` kind of types).