arduino/ArduinoCore-avr #436: Lock bit values fail verification

rockstorm101, 2021-11-14:
It's not possible to update the fuse/lock bits with the included boards.txt and recent avrdude versions.

avrdude used to mask unused bits for some microcontrollers, but this was changed some time ago.

As a result, trying to write fuse bits into an Arduino Uno leads to the following error:

```
$ /usr/bin/avrdude -v -patmega328p -cusbtiny -e -Ulock:w:0x3F:m
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x3f
avrdude: verification error; content mismatch
```

The two uppermost lock bits in the ATmega328P are unused and will always read as logic 1.

As a result, the following modification must be made to boards.txt:
```
diff --git boards.txt boards.txt
index 2dca915..d2b8dc4 100644
--- boards.txt
+++ boards.txt
@@ -67,8 +67,8 @@ uno.bootloader.tool=avrdude
 uno.bootloader.low_fuses=0xFF
 uno.bootloader.high_fuses=0xDE
 uno.bootloader.extended_fuses=0xFD
-uno.bootloader.unlock_bits=0x3F
-uno.bootloader.lock_bits=0x0F
+uno.bootloader.unlock_bits=0xFF
+uno.bootloader.lock_bits=0xCF
 uno.bootloader.file=optiboot/optiboot_atmega328.hex
 
 uno.build.mcu=atmega328p
```

This will make avrdude expect a logic 1 and prevent verification failure.

Originally reported by Gregor Riepl [[1]]. Suspected relation to #61. Please address and/or advice.

Thanks a lot.

[1]: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=985579

matthijskooijman, 2021-11-14:
I suspect this is an issue with avrdude treating reserved bits differently based on the programmer in use (for some it reads as 1, for some it reads as 0). This is patched in the version of avrdude that Arduino ships, but it seems you are using the Debian version of Arduino, which (I presume) is patched to use the stock Debian version of Arduino, which still has this bug. There should be a report about this somewhere (probably in the Arduino/Arduino or Arduino/toolchain-avr or something like that repo), but I don't have the link handy.

per1234, 2021-11-14:
> There should be a report about this somewhere

Probably either this one:
https://github.com/arduino/avrdude-build-script/issues/2
or else one of the issues linked off that thread.

rockstorm101, 2021-11-17:
Thank you both for you replies. We'll test using the non-stock avrdude version.

@matthijskooijman, looks like your patch has now been merged into avrdude's code [[1]]. Good job, thanks a lot for that.

[1]: https://savannah.nongnu.org/patch/index.php#comment2

matthijskooijman, 2021-11-18:
Hah, indeed, applied 6 days ago (after being pending for 5 years), what a coincidence :-)

Anyway, that probably makes it easier to maybe backport the patch into the Debian version, if a new release of avrude isn't coming soon (no clue about Joerg's plan's there).

Btw, your link to the merged patch was broken, you probably meant: https://savannah.nongnu.org/patch/index.php?9110#comment2

Let us know what the result of your testing is. If it works with the non-stock version, I guess this issue can be closed?