arduino/ArduinoCore-avr #437: Wire.begin() enables pullups in libraries/Wire/utility/twi.c

Ragebone, 2021-11-17:
Using Wire.begin() causes the internal pullups to be enabled in [twi.c](https://github.com/arduino/ArduinoCore-avr/blob/6507dfef3ed9af8e83acd5f02496b0ac872f3cfd/libraries/Wire/src/utility/twi.c#L88-L89).

Pullups that are 5V on a genuine Arduino.

A surprise for me when I used an Arduino Uno on a 3.3V bus. 
It luckily did not cause any apparent issues but i still believe that fact should be made more obvious.
Additionally I find it reasonable to add the option to keep those disabled. 

I hence propose to move that functionality from utility/twi.c into Wire.c and to additionally
implement a method that can keep those disabled.
How about Wire.disablePullups()?

I don't think that it's generally advisable to just outright remove or disable the Pullups by default.
There likely are projects that rely on them being enabled.

ahzahraee, 2021-12-24:
There will be no communication on the I2C bus, if the pullup resistors are disabled. The MCU will not be able to generate the I2C start condition. The MCU will not be able to transmit the I2C address. [Source](https://electronics.stackexchange.com/questions/102611/what-happens-if-i-omit-the-pullup-resistors-on-i2c-lines)

However, if you would like to use the bus at 3.3 v, you would just have to externally pull up both SDL and SDA lines to 3.3 v using strong resistors e.g 1 k ohm.

See [this](http://www.ti.com/lit/pdf/SLVA689) application report from TI for more info.

Ragebone, 2021-12-25:
@ahzahraee Thank you for your  response.

> There will be no communication on the I2C bus, if the pullup resistors are disabled. The MCU will not be able to generate the I2C start condition. The MCU will not be able to transmit the I2C [...]

Correct but only if there are no other pullups on the Bus. 
In my case, the bus was already pulled up to 3.3V and is there for operational. 

> However, if you would like to use the bus at 3.3 v, you would just have to externally pull up both SDL and SDA lines to 3.3 v using strong resistors e.g 1 k ohm.

That is sadly not true.
The Library enables pullups by default which are 5V on all genuine Arduino.
The issue is that right now there is no functionality in the Wire library to disable that beforehand. 
Meaning, i have to role my own Wire and TWI library to comment out two lines of code.
Otherwise you will have 3.3V and 5V  pullups at the same time.
Even though it did not KILL anything in my case, THAT can't be good! 
I bet that violates some specs as well. 

Another issue for me is that that isn't mentioned anywhere

My proposal is to move the code for enabling the pullups into the Wire library instead of having it in the lower level TWI lib.
Additionally there should be functionality added to the Wire lib to disable the 5V pullups before calling the Wire.begin().
That would enable the User to decide to run the Bus with external pullups only. 
Of course the default behavior should still be to enable the internal 5V pullups since there are projects that rely on those and don't have external pullups.

matthijskooijman, 2021-12-25:
> Otherwise you will have 3.3V and 5V pullups at the same time.
Even though it did not KILL anything in my case, THAT can't be good!

If the 3V3 pullup is strong enough, that should not be a real problem (it is a bit unelegant, though). For example, consider a 1k pullup to 3V3 and 20k (min value for internal IIRC) pullup to 5V, then you effectively get a 1k/20k voltage divider, so the bus voltage becomes 3.3V + (1/21*(5-3.3)) = 3.38V, which should not usually be problematic.

Ragebone, 2021-12-25:
I think i measured more then that in my case.
Around 4.?V if i remember correctly.
So i'd assume it were more like 4.7 or even 10 k Ohm. 

> edit: That doesn't mean disabling the internal pullups isn't useful, I was just commenting on this particular thing

Alls good. 
I'd amend that the OPTION to disable the pullups is the useful and missing thing.

Ragebone, 2022-01-05:
i went ahead and just quickly and lazily added that functionality on [here](https://github.com/Ragebone/ArduinoCore-avr/tree/optionalPullupsForWire)

Any feedback is welcome, i didn't test the code yet, so no PR for now.

While doing that, i noticed that ending Wire issues digitalWrite(SDx, 0).
I wonder if that shoots the bus for everyone else?
It can since i used that for some debugging, but it depended on the mode of the pins and i don't know the default. 
So maybe it does, maybe it doesn't. 
Another todo onto the list of things to check i guess.

hcwd01, 2022-01-11:
I think Ragebone is right.
bus control code is slightly inappropriately located.
existence of other devices is not considered.