arduino/ArduinoCore-avr #439: Portenta SRAM minor bug

hpssjellis, 2021-11-29:
Very strange minor bug using fixed SRAM to share data between cores.

Look for ```  Serial.println(SRAM4_START_ADDRESS, HEX); ```

If I remove it the M4 core can not update the structure and also is blocked and the blue led activated from M4 does not run.

Not really a big deal, just wondering if anyone has any idea why it is needed?

```

// Original by Khoi Hoang (Github @khoih-prog ) and Jeremy Ellis (Github @hpssjellis )
// Note: each core can be separated to it's own IDE folder

#ifdef CORE_CM7    // Start M7 Core programming

#include "mbed.h"
#include "Arduino.h"

using namespace mbed;
using namespace rtos;

//#define SRAM4_START_ADDRESS       ((uint32_t) 0x30040000)   
#define SRAM4_START_ADDRESS       ((uint32_t) 0x38000000)

struct shared_data{
  int M4toM7;
  int M7toM4;
};

int myStuff;
int localm7m4 = 0;
int localm4m7 = 0;
unsigned long myDelay = 43210;  // microSeconds not milliSeconds
unsigned long delayStart = 0;  

// Using AHB SRAM4 at 0x38000000
static struct shared_data * const xfr_ptr = (struct shared_data *) SRAM4_START_ADDRESS;

Thread M7Thread;

void setup(){
  bootM4();

  Serial.begin(115200);

  //while (!Serial);  // still blocking when removed? 
  
  // Following line is needed, not sure why? 
  Serial.println(SRAM4_START_ADDRESS, HEX); // MUST HAVE

  yield();
  
  M7Thread.start(callback(M7ThreadFunc));
  yield();
}

void M7ThreadFunc(){
  localm7m4 = 777777;
  
  while (true) {  
   if ( (micros() - delayStart ) >= myDelay) {
    delayStart = micros(); //  reset the delay time 
    yield();
    localm7m4++;
    xfr_ptr -> M7toM4 = localm7m4;
    
    Serial.print("M7 to M4: "); Serial.println(xfr_ptr -> M7toM4);
    Serial.print("M4 to M7: "); Serial.println(xfr_ptr -> M4toM7);

    Serial.println();        
    yield();
    //delay(20);
    

   //delayMicroseconds(43210);    // Note: micro-seconds not milli-seconds
  }
 }
}

void loop(){
  yield();
}

#endif

//////////////////////////////////////////////////////////

#ifdef CORE_CM4    // Start M4 Core programming

#include "mbed.h"
#include "Arduino.h"

using namespace mbed;
using namespace rtos;

//#define SRAM4_START_ADDRESS       ((uint32_t) 0x30040000)   
#define SRAM4_START_ADDRESS       ((uint32_t) 0x38000000)

struct shared_data{
  int M4toM7;
  int M7toM4;
};

int localm7m4 = 0;
int localm4m7 = 0;
unsigned long myDelay = 43210;  // microSeconds not milliSeconds
unsigned long delayStart = 0;  

static struct shared_data * const xfr_ptr = (struct shared_data *) SRAM4_START_ADDRESS;

int myStoreFromM7;

Thread M4Thread;

void setup(){
  pinMode(LEDB, OUTPUT);  // portenta blue LED
  yield();
  M4Thread.start(callback(M4ThreadFunc));
  yield();
}

void M4ThreadFunc(){
  localm4m7 = 44444444;
  
  while (true){
    if ( (micros() - delayStart ) >= myDelay) {
    delayStart = micros(); //  reset the delay time
    localm4m7++;
 
    xfr_ptr -> M4toM7 = localm4m7;

    // test if M4 an read struct changed on M7 
    if (myStoreFromM7 !=  xfr_ptr -> M7toM4){
        myStoreFromM7 =  xfr_ptr -> M7toM4;
        digitalWrite(LEDB, !digitalRead(LEDB)); // flip blue LED on and off
     }
     yield();
    // delay(20);  
   // delayMicroseconds(43210); //Note: micro-seconds not milli-seconds
  }
 }
}

void loop(){
  yield();
}

#endif

```

facchinm, 2021-11-30:
Hi Jeremy,
everything that involves communication between the cores should be safeguarded by synchronizing them (`bootM4()` just unlocks a flag so timings are more or less random). The easiest way is to just initialize (without actually using) RPC library via `RPC.begin()` on both cores, then all "random" issues should be gone :slightly_smiling_face: 
Let me know if it works!

closed 2021-11-30 by facchinm

hpssjellis, 2021-11-30:
Your right! @facchinm my working code is [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/blob/master/research/RAM-m4-m7/ram-m7-to-m4-timer-rpc-07.ino)  starting RPC solves both the weird serial.println being needed and made it so I did not have to place yield(); everywhere.

If you have time, what is RPC.begin(); doing that is needed for SRAM sharing? Is it only a few lines of code or is it a complete concept?

Thank you once again.

khoih-prog, 2021-12-01:
This is the fix without using `RPC.begin()`. Just use the `MPU_Config()`

```
// Original by Khoi Hoang (Github @khoih-prog ) and Jeremy Ellis (Github @hpssjellis )
// Note: each core can be separated to it's own IDE folder

#define STARTING_M4_VALUE         0
#define STARTING_M7_VALUE         0

#include "mbed.h"
#include "Arduino.h"

using namespace mbed;
using namespace rtos;

struct shared_data
{
  uint64_t M4toM7;
  uint64_t M7toM4;
};

#define SRAM3_START_ADDRESS       ((uint32_t) 0x30040000)
#define SRAM4_START_ADDRESS       ((uint32_t) 0x38000000)

// Using AHB SRAM4 at 0x38000000
#define SRAM_START_ADDRESS        SRAM4_START_ADDRESS

////////////////////////////////////////////////////////////////////

#ifdef CORE_CM7    // Start M7 Core programming

uint64_t localm7m4 = STARTING_M7_VALUE;

// Current 1 sec
unsigned long myDelay = 1000000;     // microSeconds not milliSeconds
unsigned long delayStart = 0;

static struct shared_data * const xfr_ptr = (struct shared_data *) SRAM_START_ADDRESS;

Thread M7Thread;

////////////////////////////////////////////

void MPU_Config()
{
  MPU_Region_InitTypeDef MPU_InitStruct;

  /* Disable the MPU */
  HAL_MPU_Disable();

  /* Configure the MPU attributes as WT for SDRAM */
  MPU_InitStruct.Enable = MPU_REGION_ENABLE;
  MPU_InitStruct.BaseAddress = D3_SRAM_BASE;
  MPU_InitStruct.AccessPermission = MPU_REGION_FULL_ACCESS;
  MPU_InitStruct.IsShareable = MPU_ACCESS_SHAREABLE;

  HAL_MPU_ConfigRegion(&MPU_InitStruct);

  /* Enable the MPU */
  HAL_MPU_Enable(MPU_PRIVILEGED_DEFAULT);
}

void setup()
{
  MPU_Config();
  bootM4();

  Serial.begin(115200);

  //while (!Serial);

  // Following issue solved by starting RPC. Not sure why.
  // Following line is needed, not sure why?
  //Serial.println(SRAM_START_ADDRESS, HEX); // MUST HAVE

  M7Thread.start(callback(M7ThreadFunc));
}

////////////////////////////////////////////

void M7ThreadFunc()
{
  static uint64_t startingMicros = micros();
  static uint64_t timestamp;

  localm7m4 = STARTING_M7_VALUE;

  while (true)
  {
    timestamp = micros();

    if ( (micros() - delayStart ) >= myDelay)
    {
      delayStart = micros(); //  reset the delay time
      
      yield();
      
      localm7m4++;
      xfr_ptr -> M7toM4 = localm7m4;

      Serial.print("M7 to M4: "); Serial.println(xfr_ptr -> M7toM4);
      Serial.print("M4 to M7: "); Serial.println(xfr_ptr -> M4toM7);

      // NUmber of M4 increment per milliseconds, around max. 1970 per ms
      // No M4 RPC                => 1980 - 2600 inc / ms
      // Both M4 and M7 RPC       => 0000 - 1970 inc / ms (slower a little bit !!!)
      Serial.print("Inc_Per_mS: "); Serial.println( ( (float) xfr_ptr -> M4toM7 - STARTING_M4_VALUE) / ( ( timestamp - startingMicros ) / 1000 ) );
      Serial.println();
    }
  }
}

////////////////////////////////////////////

void loop()
{
  // yield();
}

#endif

////////////////////////////////////////////////////////////////////

#ifdef CORE_CM4    // Start M4 Core programming

uint64_t localm4m7 = STARTING_M4_VALUE;

static struct shared_data * const xfr_ptr = (struct shared_data *) SRAM_START_ADDRESS;

uint64_t myStoreFromM7 = STARTING_M7_VALUE;

Thread M4Thread;

////////////////////////////////////////////

void setup()
{
  pinMode(LEDB, OUTPUT);  // portenta blue LED
  M4Thread.start(callback(M4ThreadFunc));
}

////////////////////////////////////////////

void M4ThreadFunc()
{
  localm4m7 = STARTING_M4_VALUE;
  
  while (true)
  {
    // without this => not increase ???
    delayMicroseconds(0);
    
    localm4m7++;

    xfr_ptr -> M4toM7 = localm4m7;

    // test if M4 an read struct changed on M7
    if (myStoreFromM7 !=  xfr_ptr -> M7toM4)
    {
      myStoreFromM7 = xfr_ptr -> M7toM4;
      digitalWrite(LEDB, !digitalRead(LEDB)); // flip blue LED on and off
    }
  }
}

////////////////////////////////////////////

void loop()
{
}

#endif
```

hpssjellis, 2021-12-01:
Thanks so much @facchinm most of my Portenta issues are done. Can you have a quick look at this weird RPC issue when using SDRAM for core communication.

[issues/38#issue here](https://github.com/arduino/ArduinoCore-mbed/issues/38#issuecomment-974119486)

hpssjellis, 2021-12-02:
Here is a very useful function to start on M7  before bootM4();
Does all the RPC.begin(); relevant functions that you mentioned.

Written by @khoih-prog 

```
void MPU_Config()
{
  MPU_Region_InitTypeDef MPU_InitStruct;

  /* Disable the MPU */
  HAL_MPU_Disable();

  /* Configure the MPU attributes as WT for SDRAM */
  MPU_InitStruct.Enable = MPU_REGION_ENABLE;
  MPU_InitStruct.BaseAddress = D3_SRAM_BASE;
  MPU_InitStruct.AccessPermission = MPU_REGION_FULL_ACCESS;
  MPU_InitStruct.IsShareable = MPU_ACCESS_SHAREABLE;

  HAL_MPU_ConfigRegion(&MPU_InitStruct);

  /* Enable the MPU */
  HAL_MPU_Enable(MPU_PRIVILEGED_DEFAULT);
}

```