arduino/ArduinoCore-avr #445: Request adding Serial.writeIdle()

microPaul, 2021-12-23:
Request adding Serial.writeIdle() command; 
returns 1 when no data remain in Serial TX buffer and the UART TX is idle, 
returns 0 if not all serial data have been completely transmitted.

I don't see any function like this in the Serial reference ( https://www.arduino.cc/reference/en/language/functions/communication/serial/ ).  

My goal for this function is to provide a non-blocking means to know when all the data I have sent to the TX UART have been completely transmitted by the TX UART, so that I can safely call Serial.end() and avoid prematurely stopping serial transmission.  I'm using the UART TX output to control a low power RF transmitter (transmitter is active on HIGH control input) and want a way to know when I can safely turn off the UART TX output (and thus ensure that the transmitter is off).

ahzahraee, 2021-12-24:
You just have to use another digital pin for controlling your RF transmitter.

microPaul, 2021-12-24:
> You just have to use another digital pin for controlling your RF transmitter.

I don't think that really addresses the issue.  The issue is primarily one of knowing when it's safe to disable the transmitter, not the means of actually disabling the transmitter.

The same problem applies if an external multiplexer was added to the TX output of the UART, to route the serial output to different independent devices.  One needs to know for certain when the serial transmission is complete before changing the pathways within the multiplexer.

ahzahraee, 2021-12-24:
ok I just checked the datasheet for atmega 168/328 family:

> 20.6.5 Disabling the Transmitter
> The disabling of the Transmitter (setting the TXEN to zero) will not become effective until ongoing and pending
> transmissions are completed, i.e., when the Transmit Shift Register and Transmit Buffer Register do not contain
> data to be transmitted. When disabled, the Transmitter will no longer override the TxDn pin.

This probably holds true for other atmega MCU, but I have not checked.

The code for Serial.end() works by seeting TXEN to zero. So, you should be find calling Serial.end() as it will wait for transmission to end.

microPaul, 2021-12-24:
> So, you should be find calling Serial.end() as it will wait for transmission to end.

Yes, understood, but I'd prefer something that was non-blocking.