arduino/ArduinoCore-avr #451: Wire.endTransmission(false)

Palingenesis, 2022-01-28:
New at using GitHub so apologies if I am doing this wrong.

Using the Wire Library.
Wire.endTransmission(false);
As I understand it this command should not send a STOP "P" after sending/attempting_to_send data in the current transition.
I have visited Arduino Forums to try and find an answer with no joy.
Please take a look, it will explain what my issue is. [Wire Library [repeated start] ](https://forum.arduino.cc/t/wire-library-repeated-start/952211)

What I would also like to achieve is what is written in the [i2c_bus_specification_1995.pdf](https://www.i2c-bus.org/fileadmin/ftp/i2c_bus_specification_1995.pdf)
Section **9.1.2 START byte**.

I don't think the Wire.endTransmission(false);  is working correctly, or perhaps needs another overload.
May be we need: Wire.endTransmission(Boolean, Boolean); first Boolean = Stop bit, second Boolean = Dummy ACK.

Ragebone, 2022-03-05:
As far as i have understood your issue on the Arduino Forum, your Slave is breaking the i2c spec.
Additionally i think that the STOP condition is dependent on the previous NACK.
Since your Slave does not ACK, a STOP is the expected and i'd say correct behavior.
So the issue is in my opinion not with the Wire lib and this specific function but with your specific slave device and implementation.

But you can tweak the i2c statemachine on the Master to break spec and hopefully do what you want it to do.
You can find the default code [here](https://github.com/arduino/ArduinoCore-avr/blob/6507dfef3ed9af8e83acd5f02496b0ac872f3cfd/libraries/Wire/src/utility/twi.c#L506-L566)
If you are using other Master devices like an ESP8266 or ESP32, that situation looks a bit different.

Ragebone, 2022-03-05:
Have a look at [this](https://github.com/arduino/ArduinoCore-avr/blob/6507dfef3ed9af8e83acd5f02496b0ac872f3cfd/libraries/Wire/src/utility/twi.c#L506-L566)

Palingenesis, 2022-03-06:
I do apologise for not closing this issue earlier.
I closed [Wire Library [repeated start]](https://forum.arduino.cc/t/wire-library-repeated-start/952211) after achieving a solution.
I have found that the protocol used by the device I was trying to communicate with, is a bespoke version I2C.
I have managed to use and edit a soft version of Arduino I2C Library to communicate with the device.
[Control Cybot from Web Browser using ESP32-CAM](https://tims-arduino-cybot.blogspot.com/2022/02/control-cybot-from-web-browser-using.html)

closed 2022-03-06 by Palingenesis

Ragebone, 2022-03-07:
Hey awesome that you got it working for your project.
please excuse my commenting twice and then deleting them shortly after. 
I made the mistake of assuming things and not reading the thread to its end.

Now with that out of the way, i am still not convinced that the described behavior is within or outside of the standard.
My initial thought was that it definitively is not within spec but i'm not sure anymore.
Main reason for that is that i could not yet find a statement that a STOP has to follow on a NACK.
The obvious and totally reasonable implementation is what is currently the case with a STOP following a NACK immediately.  

But what if you want to have one transaction where you address multiple different slave devices and read or write data to them?
At least that is a thing i remember from tracing the SMBus activity of an BMC on a motherboard.
But that might currently work as long as none of the addressed slaves unexpectedly NACK, which they shouldn't?

Another thing i find weird is that i don't think there should be a difference in your observed behavior between having STOP conditions and having none.
The only way i can get myself to make sense of it is that the controllers protocol expects to be addressed exactly or at least twice within a single transaction with only the second being ACKed?

Weirdness for sure. 
Maybe that was meant as a protection feature to prevent being found on a bus by scanning and dumping?
Well, what ever.

Palingenesis, 2022-03-07:
I got a bit lost on the way to which is right and which wrong.
When I first looked at it, having the option to not send a stop made sense in that: If the programmer used this option, the programmer had full control on when the bus is released to other devices connected to the bus. The programmer would have to add some sort of timeout in there code to release the bus to other devices, whether a NACK is invoked or not.
But when I looked at the signals with the data logger, when the master invoked a NACK if no reply was received, in doing so it added a STOP.
If a STOP has been sent, the data logger interpreted the next START as a normal START not a RE-START. Also if a STOP has been sent the bus is open to any other device on the bus.

This is how I saw it at the start, My brain crashed towards the end, so I just modified a soft library so that if the option to not send a stop was used, it never sent a stop until I wanted it to.
There was some other tweaks to set the condition of the two wires.

I was not able to modify the Hard Wired I2C library because you only have access to the address of registers that have set tasks, I was not able to work out how to individually change the state of the two wires.