arduino/ArduinoCore-avr #458: Release 1.8.5 Download missing

NicoHood, 2022-02-17:
I was expecting to have the lastest release available here:
https://downloads.arduino.cc/cores/avr-1.8.5.tar.bz2

Maybe another compression was used, can you tell me which? Or where to find those files? I'd use the files from github, but those are not signed.

per1234, 2022-02-20:
Hi @NicoHood. Although the 1.8.5 release has been created here on GitHub, it has not yet been added to the production package index for distribution via the Arduino Boards Manager.

Once that has been done, I'll update this issue and you'll then be able to get the archive URL from the `packages[*].platforms[*].url` field of the `arduino:avr@1.8.5` release from https://downloads.arduino.cc/packages/package_staging_index.json. For example, this is the relevant section of the package index for the 1.8.4 release:

```json
{
  "name": "Arduino AVR Boards",
  "architecture": "avr",
  "version": "1.8.4",
  "category": "Arduino",
  "help": {
    "online": "http://www.arduino.cc/en/Reference/HomePage"
  },
  "url": "http://downloads.arduino.cc/cores/staging/avr-1.8.4.tar.bz2",
```

>Maybe another compression was used, can you tell me which?

It will be `.tar.bz2` just like the previous ones.

facchinm, 2022-02-21:
Core published in production, no need to use the `staging` json anymore :wink:

NicoHood, 2022-02-22:
The gpg signature is not available: https://downloads.arduino.cc/cores/avr-1.8.5.tar.bz2.asc

per1234, 2022-12-28:
I'm closing this in favor of the dedicated issue tracking the absence of GPG signatures for the recent releases (thanks @M-Reimer): https://github.com/arduino/ArduinoCore-avr/issues/519

closed 2022-12-28 by per1234