arduino/ArduinoCore-avr #461: IPAddress.h defines variable INADDR_NONE

pianistrevor, 2022-02-22:
The header file has a line:
`const IPAddress INADDR_NONE(0,0,0,0);`
Which means every file that includes this header will be assigned its own INADDR_NONE. Projects with limited static memory will encounter problems when this file is included in many .cpp files.

A suggested workaround (which I have confirmed works) would be to make IPAddress.h say `extern const IPAddress INADDR_NONE;` and define the actual variable in IPAddress.cpp as `const IPAddress INADDR_NONE(0,0,0,0);`

per1234, 2022-02-22:
Thanks for your suggestion @pianistrevor!

There was a previous report of this https://github.com/arduino/Arduino/issues/1007

It was considered resolved by doing just what you suggested in the `arduino/ArduinoCore-API` repository:
https://github.com/arduino/ArduinoCore-API/commit/3dbd7d7d338ff2d93c8b9dd8c833967fdb69b7c6

However, the "Arduino AVR Boards" core has not yet transitioned to using [ArduinoCore-API](https://github.com/arduino/ArduinoCore-API)(https://github.com/arduino/ArduinoCore-avr/pull/329), so the issue persists in this core.

pianistrevor, 2022-02-22:
Good to know, thank you! Will this be something this core does in the future, or should I plan to configure a different board? I’m currently using a CI framework that installs AVR...