arduino/ArduinoCore-avr #466: USART define warnings - U3Xn in HardwareSerial_private.h

mikrocoder, 2022-03-03:
Hello,

there is an error in the USART define links. A bit definition called **U3Xn** does not exist. This is surely a typo and must be **U2Xn**. n stands for USART 0...3. If you activate the compiler flag -Wundef you get the following output. Can someone please correct this in HardwareSerial_private.h?
Thank you.

```cpp
In file included from C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial.cpp:33:0:
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:29: warning: "U3X3" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                             ^~~~
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:37: warning: "U3X0" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                                     ^~~~
In file included from C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial2.cpp:27:0:
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:29: warning: "U3X3" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                             ^~~~
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:37: warning: "U3X0" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                                     ^~~~
In file included from C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial0.cpp:27:0:
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:29: warning: "U3X3" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                             ^~~~
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:37: warning: "U3X0" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                                     ^~~~
In file included from C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial1.cpp:27:0:
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:29: warning: "U3X3" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                             ^~~~
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:37: warning: "U3X0" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                                     ^~~~
In file included from C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial3.cpp:27:0:
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:29: warning: "U3X3" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                             ^~~~
C:\Arduino IDE Portable\Mega2560\arduino-1.8.19\portable\packages\arduino\hardware\avr\1.8.5\cores\arduino\HardwareSerial_private.h:80:37: warning: "U3X0" is not defined, evaluates to 0 [-Wundef]
         UDRIE3 != UDRIE0 || U3X3 != U3X0 || UPE3 != UPE0 || \
                                     ^~~~
```

mikrocoder, 2022-03-04:
[Link to the proposed change](https://github.com/arduino/ArduinoCore-avr/compare/master...mikrocoder:patch-1)

I did not want to start a duplicate request now, therefore the link. I hope this makes your work easier.

CombiesGit, 2022-12-08:
After update from 1.8.5 to 1.8.6
The problem still persists.