arduino/ArduinoCore-avr #501: Compiling fails for arduino-usbserial

kbastronomics, 2022-10-03:
Hi,
I searched for any issue with this before creating this issue and did not find any.
I am in need to use a custom PID/VID plus Product and Manufacture strings.

This issue compiling is happening ever with original unchanged source for arduino-usbserial.
I copied arduino-usbserial to the lufa/Projects as directed and did a make clean;make and everything compiles until it hits arduino-usbserial.  then it dies.

I checked paths and things are there,  this is likely a simple issue but I'm at a loss

I did remove arduino-usbserial and LUFA compiles without error.   this only happens when added arduino-usbserial to the Project directory.

This is the last part of the compile before it fails.
is the issue with the latest LUFA?

make[1]: Entering directory '/home/sandra/Arduino/lufa/Projects'
make[2]: Entering directory '/home/sandra/Arduino/lufa/Projects/arduino-usbserial'
../../LUFA/makefile:38: Build/DMBS/DMBS/doxygen.mk: No such file or directory
make[2]: *** No rule to make target 'Build/DMBS/DMBS/doxygen.mk'.  Stop.
make[2]: Leaving directory '/home/sandra/Arduino/lufa/Projects/arduino-usbserial'
make[1]: *** [makefile:33: arduino-usbserial/] Error 2
make[1]: Leaving directory '/home/sandra/Arduino/lufa/Projects'
make: *** [makefile:22: all] Error 2

kbastronomics, 2023-04-13:
no,  never got it to work

per1234, 2025-03-08:
LUFA's build system has a dependency on [**doxygen**](https://www.doxygen.nl/). The problem is caused by having a variant of **doxygen** installed that is not compatible with the LUFA build system:

https://github.com/abcminiuser/lufa/issues/90

Since the problem is not directly related to any of the content of this repository, I'll close the issue as off topic.

closed 2025-03-08 by per1234