arduino/ArduinoCore-avr #508: Eliminate compiler warnings for unused parameter.

habazut, 2022-10-25:
```
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp: In function 'void* operator new(std::size_t, std::nothrow_t)':
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp:59:60: warning: unused parameter 'tag' [-Wunused-parameter]
 void * operator new(std::size_t size, const std::nothrow_t tag) noexcept {
                                                            ^~~
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp: In function 'void* operator new [](std::size_t, const std::nothrow_t&)':
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp:68:63: warning: unused parameter 'tag' [-Wunused-parameter]
 void * operator new[](std::size_t size, const std::nothrow_t& tag) noexcept {
                                                               ^~~
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp: In function 'void operator delete(void*, const std::nothrow_t&)':
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp:103:55: warning: unused parameter 'tag' [-Wunused-parameter]
 void operator delete(void* ptr, const std::nothrow_t& tag) noexcept {
                                                       ^~~
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp: In function 'void operator delete [](void*, const std::nothrow_t&)':
/home/haba/.arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/new.cpp:106:57: warning: unused parameter 'tag' [-Wunused-parameter]
 void operator delete[](void* ptr, const std::nothrow_t& tag) noexcept {
 ```
 Just fix. with a `void (tag);` With -Werror it won't compile at all.
 Regards,
 Harald.

matthijskooijman, 2022-10-26:
There are already a number of PRs for this, I would prefer merging #487 to fix this.

noisymime, 2024-08-11:
Is there any hope of getting one of the multiple PRs that resolve this annoyance pulled in? 

The PRs are very low impact/risk, it really would be nice to have this cleaned up.