arduino/ArduinoCore-avr #510: Fixed-width integer type aliases are leaked from `USBAPI.h`

niculaionut, 2022-11-05:
It seems inappropriate that `u8`, `u16`, and `u32` are leaked from `USBAPI.h`, which in turn is included in `Arduino.h`, and thus polluting the global namespace:

https://github.com/arduino/ArduinoCore-avr/blob/42fa4a1ea1b1b11d1cc0a60298e529d37f9d14bd/cores/arduino/USBAPI.h#L29-L31

If you try to define `u16` yourself as `using u16 = uint16_t;` while including `Arduino.h`, you'll get a compilation error because this alias conflicts with the one from `USBAPI.h`.

matthijskooijman, 2022-11-05:
Sounds wrong to me, I'd be in favor of removing them. Especially since (at first glance) it seems the header file does not actually use these types at all either.

yuriymakhin, 2025-12-29:
Yes please.