arduino/ArduinoCore-avr #513: Arduino nano every won't work as I2C Master with address

Schaumi19, 2022-11-18:
The arduino nano every won't work as I2C Master, if you give it an address. For example: Wire.begin(4);
Every thing works fine without address.
It took me a long time to figure that out.
I don't think thats on purpose.
I have been using the latest IDE, so thats not the Problem.

I'm not completely sure, but I think it crashes at: Wire.endTransmission();

### Temporary Fix:
1. End I2C connection and start without address
2. Use as master
3. End I2C connection and start with address
4. Use as slave

per1234, 2022-11-18:
Hi @Schaumi19. Thanks for your report.

I think this is the intended design of the library. It is specified in the documentation:

https://www.arduino.cc/reference/en/language/functions/communication/wire/begin/

> ### Parameters
> 
> `address`: the 7-bit slave address (optional); if not specified, join the bus as a controller device.

closed 2022-11-18 by per1234

Schaumi19, 2022-11-19:
Hi @per1234. Thanks for your fast reply.
I understand your point, but I don't think so because:
* I2C is a multi master bus, so it is kind of necessary to have an address, if there are multiple masters(one at a time) in one system.
* In all examples i have found it says: ` Wire.begin(); // join i2c bus (address optional for master) `
* It works fine on all other Arduinos I have testet, like the **nano, uno and mega**