arduino/ArduinoCore-avr #518: Version 1.8.6 missing on downloads.arduino.cc, no signature for version 1.8.5

M-Reimer, 2022-12-28:
Hello Arduino team. I was just trying to build a new AVR core for Arch Linux. But just swapping the version number, as usually works to get a new version packaged, didn't do it this time as the download is missing on "downloads.arduino.cc":

https://downloads.arduino.cc/cores/avr-1.8.6.tar.bz2

And while at it: The GPG signature for version 1.8.5 is missing. Can you perhaps sign the new 1.8.6 version? Would make it probably easier to get this finally updated upstream on Arch.

M-Reimer, 2022-12-28:
The GPG signature part for 1.8.5 is already covered here: https://github.com/arduino/ArduinoCore-avr/issues/458

So maybe take this as a reminder that, if you upload 1.8.6 to the download server, that you also create a GPG signature for the archive file.

per1234, 2022-12-28:
Hi @M-Reimer.

>the download is missing on "downloads.arduino.cc"

This is an incorrect conclusion. The platform archive file is present on `downloads.arduino.cc`. The problem is that it is under a different path on that domain compared to previous releases:

http://downloads.arduino.cc/cores/staging/avr-1.8.6.tar.bz2

You can find the URL of any of the platform archives by checking the package index:

https://downloads.arduino.cc/packages/package_index.json

```json
        {
          "name": "Arduino AVR Boards",
          "architecture": "avr",
          "version": "1.8.6",
          "category": "Arduino",
          "help": {
            "online": "http://www.arduino.cc/en/Reference/HomePage"
          },
          "url": "http://downloads.arduino.cc/cores/staging/avr-1.8.6.tar.bz2",
          "archiveFileName": "avr-1.8.6.tar.bz2",
          "checksum": "SHA-256:ff1d17274b5a952f172074bd36c3924336baefded0232e10982f8999c2f7c3b6",
          "size": "7127080",
```

It is not clear to me whether this "staging" path was intended, since the directory name implies an ephemeral status that is not appropriate for releases in the production package index. So we can use this issue to track that question even though the actual situation is not quite as described in the initial report.

> And while at it: The GPG signature for version 1.8.5 is missing. Can you perhaps sign the new 1.8.6 version?

Please open a dedicated issue about that. You can see from the previous report at https://github.com/arduino/ArduinoCore-avr/issues/458 that tacking it on as an afterthought to another report that is only tangentially related doesn't get good results.

closed 2022-12-28 by M-Reimer