arduino/ArduinoCore-avr #534: -DCDC_DISABLED not working

Phoen11xx, 2023-05-15:
Hello. I am trying to create a usb device and I need to disable the CDC so that only the hid device works.
To install the code, I use usbasp. After enabling the -DC DC_DISABLED parameter or adding define, the CRC still works and I get a hid device plus a non-working device.
Does anyone know the solution to this problem?

per1234, 2023-05-15:
Hi @Phoen11xx. Thanks for your report. I know there has been a recent discussion about the use of this macro on Arduino Forum:

https://forum.arduino.cc/t/remove-serial-port/1123859

Is that you? I ask because that discussion contains valuable information about how the users are setting the macro (via the boards definition instead of trying to do it via the sketch code).

Phoen11xx, 2023-05-15:
These not my forum thread. but i have exactly the same problem.

facchinm, 2023-05-15:
The issue is very likely due to how Windows caches the USB descriptors; you can try changing the productVersion field [here](https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/USBCore.cpp#L82)  , replacing `0x100` with any other number, to check if it forgets the previous association.

Phoen11xx, 2023-05-15:
I recorded a video showing this problem:
https://www.youtube.com/watch?v=9o1CELyCOsc
It doesn't look like the problem is in the cache. digitalWrite still working but mouse.move not working with CDC_DISABLED define

Phoen11xx, 2023-05-15:
@facchinm yes I was wrong about the cache. I change `0x100` to `0x101` and it worked