arduino/ArduinoCore-avr #549: Wire library read() returns 0xFF on reading only one byte

velechit, 2023-12-15:
I am having a wierd problem with my i2c code. The problem is that when I try to read only 1 byte in the master I get 0xFF but if I read 2 bytes I get proper value of the first byte.

Wire.requestFrom(SLAVE_ADDRESS,1);
data=Wire.read();

=> gives data as 0xFF

Wire.requestFrom(SLAVE_ADDRESS,2);
data0=Wire.read();
data1=Wire.read();

=> gives data0 as 0xA5 (which is expected from the single byte read) 

Am I doing anything wrong?

zerovijay, 2024-01-29:
can i get more information about the issue

ItsMe6666, 2024-06-03:
Wire.requestFrom(SLAVE_ADDRESS, 1);
while (Wire.available()) { // slave may send less than requested
    data0  = Wire.read();
    Serial.print(data0 ); 
  }