arduino/ArduinoCore-avr #554: Debug optimization incorrectly set to -Os should be -O0

NormanDunbar, 2024-02-24:
### Describe the problem

When debugging an Uno, I've noticed that there are still optimizations taking place even though I've checked the "Optimize for debugging" menu item. DIsassembling the generated elf file with avr-objdump shows "weird" output. Sometimes complete function are missing, but the code still works.

Looking at a verbose compile, I see that the optimization setting for the compiler and linker are set to `-Os` which is optimizing for size, and not `-O0` which it should be. (I think!)

I have compiled with the correct (!) setting and all is well. I can disassemble the code and step through it (using simulavr) correctly, which is not always possible with the current optimization.

Cheers,
Norman Dunbar (Author: Arduino Software Internals and Arduino Interrupts)

### To reproduce

Compile with debug optimization and step through the code in simulavr. Or disassemble the elf file with avr-objdump.

### Expected behavior

Correct debug optimization settings.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

LInux Mint 21.3 64 bit.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-02-24:
Hi @NormanDunbar. The platform developer has complete control over the effect (or lack thereof) of the IDE's **Sketch > Optimize for Debugging** option through the platform configuration files:

https://arduino.github.io/arduino-cli/latest/platform-specification/#optimization-level-for-debugging

The `compiler.optimization_flags.debug` property has not been configured in the "Arduino AVR Boards" platform's configuration files. This means the **Sketch > Optimize for Debugging** option in Arduino IDE has no effect when compiling for a board of this platform. The compilation will proceed as appropriate for production application regardless of the setting of that menu option.

Since the platform has not been configured for use with the [Arduino IDE 2.x integrated sketch debugger](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-debugger), a `compiler.optimization_flags.debug` property in this platform would not be directly useful to Arduino IDE users. It could be for users like yourself who are using an external debugger though so we can consider this a feature request. We must weigh the benefit for the advanced users with this use case against the harm such a configuration will cause for the beginners who will enable the option without understanding what it does, forget all about it, and then suffer from excessive program size from there on.

NormanDunbar, 2024-02-25:
Good Morning @per1234 .

Many thanks for a prompt response. If I'm reading you correctly, I think you are saying that this debug optimisation flag has no effect on the Uno as debugging as a whole is not implemented?

* I tried adding `compiler.optimization_flags.debug=-O0 -g` to `platform.local.txt` and after restarting the IDE, this had no effect.
* I also tried adding it to `platform.txt` and again, this had no effect.

I wasn't sure from your reply if you simply meant that this setting hadn't been configured, but it seems that it's not actually available to be used.

No worries, I can compile and link manually with the proper debug settings.

Best regards,
Norman.

per1234, 2024-02-25:
> I think you are saying that this debug optimisation flag has no effect on the Uno as debugging as a whole is not implemented?

No, I was saying that the developers likely saw no reason to configure the platform to adjust the compilation commands according to the "Optimize for Debugging" setting since the primary use case for that setting is the Arduino sketch debugger and this platform can't be used with the Arduino sketch debugger.

However, the "Optimize for Debugging" feature can be used by a platform even if it has not been configured for sketch debugger support.

> this had no effect

Did you add a `compiler.optimization_flags` property reference to the compilation command patterns?

The patch should look something like this:

```patch
--- a/platform.txt
+++ b/platform.txt
@@ -17,15 +17,21 @@ compiler.warning_flags.default=
 compiler.warning_flags.more=-Wall
 compiler.warning_flags.all=-Wall -Wextra

+# Fallback property definition for compatibility with development tools that don't have the "Optimize for Debugging" control.
+compiler.optimization_flags=-Os
+# See: https://arduino.github.io/arduino-cli/latest/platform-specification/#optimization-level-for-debugging
+compiler.optimization_flags.release=-Os
+compiler.optimization_flags.debug=-Og -g3
+
 # Default "compiler.path" is correct, change only if you want to override the initial value
 compiler.path={runtime.tools.avr-gcc.path}/bin/
 compiler.c.cmd=avr-gcc
-compiler.c.flags=-c -g -Os {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects
-compiler.c.elf.flags={compiler.warning_flags} -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections
+compiler.c.flags=-c -g {compiler.optimization_flags} {compiler.warning_flags} -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects
+compiler.c.elf.flags={compiler.warning_flags} {compiler.optimization_flags} -g -flto -fuse-linker-plugin -Wl,--gc-sections
 compiler.c.elf.cmd=avr-gcc
 compiler.S.flags=-c -g -x assembler-with-cpp -flto -MMD
 compiler.cpp.cmd=avr-g++
-compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto
+compiler.cpp.flags=-c -g {compiler.optimization_flags} {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto
 compiler.ar.cmd=avr-gcc-ar
 compiler.ar.flags=rcs
 compiler.objcopy.cmd=avr-objcopy
```

You can do this in `platform.local.txt` if you like.

NormanDunbar, 2024-02-25:
Hi @per1234 ,

oops! What a plank I am. I completely forgot to add the property reference to the compilation references. :cry:  I have done so, as per your example, and everything is fine now. I have tested with an optimised for debugging and a release compilation and the debug settings are perfect.

Thank you very much indeed. This is a great help. I can get back to debugging code for my next Arduino book!

I've now closed this issue. Thanks again.

All the best,
Norman. (Author of Arduino Software Internals and Arduino Interrupts)

closed 2024-02-25 by NormanDunbar