arduino/ArduinoCore-avr #567: Update to Arduino Nano Every breaks FastLEd compile

zackees, 2024-09-13:
Not sure what's going on but users are reporting that a common pin (?) no longer works with one of the recent update:

https://github.com/FastLED/FastLED/actions/workflows/build_nano_every.yml

per1234, 2024-09-13:
Hi @zackees. First of all, this is not the repository for the Nano Every's Arduino boards platform. So reports related to Nano Every are off topic here.

Second of all, you haven't presented anything that would lead me to believe that this is a problem with the code maintained by Arduino rather than a bug in the FastLED code.

I don't know how you do things in the FastLED project, but we hold bug reports to a far higher quality standard in Arduino's repositories. This issue is low quality even for a beginner, but I would expect much more from someone with as much experience as you. Please make a stronger effort if you submit other issues to Arduino's repositories in the future.

Less structured discussion is always welcome over on Arduino Forum:

https://forum.arduino.cc/

closed 2024-09-13 by per1234

zackees, 2024-09-13:
Reverting arduino core fixes the issue. This broke with your arduino 3 update.

https://github.com/FastLED/FastLED/issues/1638#issuecomment-2211954247

I was hoping this bug report could start a discussion about generating a repro case. Instead you are trying to qualify me. Bizarre.

If you hold bug reports in such high standard then it's up to you to provide a bug report template to state your requirements, which you don't.

zackees, 2024-09-13:
And why isn't this the place to report this? The Arduino Nano is an avr board and this is the arduino avr core repo.