arduino/ArduinoCore-avr #586: twi_readFrom() does not proerly handle timeout

Ztirom45, 2024-12-29:
### My program freezes randomly. After debugging I found out this: 

gdb output:
```
Program received signal SIGINT, Interrupt.
0x00001ad8 in twi_readFrom (address=address@entry=104 'h', 
    data=0x8009d0 <TwoWire::rxBuffer> <incomplete sequence \302>, 
    length=length@entry=1 '\001', sendStop=sendStop@entry=1 '\001')
    at /home/moritz/.platformio/packages/framework-arduino-avr/libraries/Wire/src/utility/twi.c:219
219         if((twi_timeout_us > 0ul) && ((micros() - startMicros) > twi_timeout_us)) {
```
```
(gdb) print twi_timeout_us
$1 = 0
```
```
(gdb) print startMicros
$2 = 763126656
```
using:
Adafruit_ADXL345_U.h: Adafruit_MPU6050.getEvent()

I am not sure how the timeout handling should work, but I don't think that it should freeze.

matthijskooijman, 2024-12-30:
Did you enable the timeout with setWireTimeout? https://reference.arduino.cc/reference/en/language/functions/communication/wire/setwiretimeout/

Ztirom45, 2024-12-30:
Thank you for your quick reply.
No, I think Adafruit_ADXL345_U and the underlying  Adafruit BusIO don't use setWireTimeout. I will ask there.

closed 2024-12-30 by Ztirom45